google-research/circuit_training #3: Support LEF/DEF

tfboyd, 2022-01-19:
Please add a comment if you need/want LEF/DEF support so we can prioritize asks.

QuantamHD, 2022-01-19:
There's a python API for openDB that supports LEF and DEF writing.

QuantamHD, 2022-01-19:
@tfboyd I work inside Google (ethanmoon@) as well on open source hardware EDA tools. Would love to discuss how to integrate this into some of our work.

thesukantadey, 2022-01-19:
@tfboyd We are a hardware security research group at University of Florida. We need LEF/DEF support to utilize CircuitTraining for our research.

Yangxiaojun1230, 2022-03-21:
@tfboyd I am doing research into floorplan, I need LEF/DEF support to validate the CircuitTraining on other open source datasets.

tfboyd, 2022-03-21:
We are having a similar issue internally.  I realize that sounds absurd.  Our current process exports from another tool and is never reading LEF/DEF.  I wish I had an ETA. We currently do not have a person assigned. I do appreciate this ping because it helps me try in increase the priority even higher.

I am linking this entire thread into our weekly sync.

Yangxiaojun1230, 2022-04-17:
> We are having a similar issue internally. I realize that sounds absurd. Our current process exports from another tool and is never reading LEF/DEF. I wish I had an ETA. We currently do not have a [...]

Hi author,
Do you have any update ? I implemented the LEF/DEF extractor in python.
So far , the extractor realized in two steps: 1. Extract macros , std cells, pins and generate hypergraph data format for hmetis input. 2. Collect macros , clusters and pins and generate initial.plc and netlist.pb.txt
But I don't know how to define the “x” and "y"  of the cluster and the "weight" of the input. Could you offer a simple dataset to verify the extractor? Thanks

ZhiangWang033, 2022-06-21:
Hi everyone,

This is a repo (https://github.com/TILOS-AI-Institute/MacroPlacement/tree/main/CodeElements) discussed the details about the the "x" and "y" of the cluster.  They also provide the LEF/DEF translators.

thesukantadey, 2022-06-21:
@ZhiangWang033, thanks for sharing it.

tfboyd, 2022-07-11:
We have added the [grouping (clustering) code](https://github.com/google-research/circuit_training/tree/main/circuit_training/grouping). This does require the netlist to already by in the protobuff structure which is documented [here](https://github.com/google-research/circuit_training/blob/main/docs/NETLIST_FORMAT.md).  We are working with the third-party vendor to allow us to release the tcl script we use to export from the tool we use internally to the protobuf. That discussion is taking longer than expected. We do not directly use LEF/DEF internally for this flow or we would have open sourced that path already.

braamBeresford, 2022-07-14:
@ZhiangWang033 or @thesukantadey  did you successfully run the LEF/DEF translators? 
The code seems stale.

tfboyd, 2022-10-21:
It was a long time coming and a big thank you to TILOS-AI-Institute.

[TILOS-AI-Institute](https://www.tilos.ai/) has created a [script](https://github.com/TILOS-AI-Institute/MacroPlacement/tree/main/CodeElements/FormatTranslators) to convert LEF/DEF and Bookshelf to the [Netlist Protocol Buffer](https://github.com/google-research/circuit_training/blob/main/docs/NETLIST_FORMAT.md) used as the input for circuit training.

closed 2022-10-21 by tfboyd