google-research/circuit_training #8: SHA256 optimized asic

stevedana, 2022-01-27:
Do you recommend this framework for designing an ASIC that is optimized for sha256?

Related question: In the paper accompanying this framework, it's stated that "We group millions of standard cells into a few thousand clusters using hMETIS (Karypis & Kumar, 1998)". I was wondering if this is too coarse of an approximation for optimizing SHA256 specifically. 

Appreciate your help, thank you

tfboyd, 2022-01-28:
@esonghori   Can you answer this or forward it to someone else?

esonghori, 2022-01-28:
Hi, the circuit training is designed to work on any macro floorplaning task regardless of the functionality of the block.  

The coarse clustering is needed to improve the runtime. The number of clusters is a hyperparameter that controls the trade-off between the speed and the correlation of our approximate placement quality with the real quality (i.e., placement quality after the detailed routing).  We will include the code for clustering and calling hMETIS in future versions. You can tune the number of clusters according to your needs. A few thousand clusters worked well for us for floorplaning TPU blocks.

closed 2022-01-28 by esonghori