google-research/circuit_training #10: Continuing the same thread as in issue #6

Mrinal18, 2022-02-06:
Continuing the same thread as in #6 

Hello, @tfboyd thank you so much for the information, can you also suggest from where you extracted the dataset/macros? because Ariane RISC-V open-source Github link is outdated (According to this https://github.com/riscv-software-src/riscv-tools/issues/333)

Thanks,
Mrinal

Yangxiaojun1230, 2022-02-22:
> 

> Continuing the same thread as in #6
> 
> Hello, @tfboyd thank you so much for the information, can you also suggest from where you extracted the dataset/macros? because Ariane RISC-V open-source Github link is outdated (According to this [riscv-software-src/riscv-tools#333](https://github.com/riscv-software-src/riscv-tools/issues/333))
> 
> Thanks, Mrinal

Hi Mrinal,
Do you find a solution to transform the verilog to the netlist file ?

esonghori, 2022-03-20:
Hi Mrinal, 

There are a few steps from a Verilog code to a netlist format readable by the project.

1. Synthesizing the netlist.
2. Providing the physical information including macro sizes, the macro pin locations (relative to the center of macro), block size, and I/O port locations. 
3. Converting the logical connections and physical information into a special `tensorflow.GraphDef` [protobuffer](https://developers.google.com/protocol-buffers) format[^1]. 
4. [Optional] Clustering the standard cell to reduce the runtime and memory footprint.

Internally, we rely on commercial synthesis tool used in production for Step 1 and 2. Step 3 also depends on the tool, and as a result, we can't readily open-source them due to IP and trade secret issues.
We plan to have an open-source script to translate LEF/DEF format which you can get from synthesis tools to `tensorflow.GraphDef` proto to resolve this problem (see #3).

For Step 4, as mentioned in the paper, we use hMETIS, an open-source graph partitioning tool.

[^1]: The format description is not open-sourced yet (see #12), but you can take a look at [environment/test_data/ariane/netlist.pb.txt](https://github.com/google-research/circuit_training/blob/main/circuit_training/environment/test_data/ariane/netlist.pb.txt) as an example to figure it out.

closed 2022-03-20 by esonghori