google-research/circuit_training #9: Issues while running Docker image

Mrinal18, 2022-02-06:
I am getting errors when running docker 
[image]

I used the run commands from here: https://github.com/google-research/circuit_training/blob/main/docs/ARIANE.md#docker

`$ docker build --tag circuit_training:core -f tools/docker/ubuntu_circuit_training tools/docker/
`
Full message : 

```
=> ERROR [9/9] RUN python3 -mpip --no-cache-dir install tf-agents[reverb]                               7.2s
------                                                                                                        
 > [9/9] RUN python3 -mpip --no-cache-dir install tf-agents[reverb]:                                          
#12 0.492 Collecting tf-agents[reverb]                                                                        
#12 0.755   Downloading tf_agents-0.11.0-py3-none-any.whl (1.3 MB)                                            
#12 0.896      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.3/1.3 MB 9.3 MB/s eta 0:00:00                       
#12 0.959 Collecting wrapt>=1.11.1                                                                            
#12 1.004   Downloading wrapt-1.13.3.tar.gz (48 kB)
#12 1.005      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 48.9/48.9 KB 232.0 MB/s eta 0:00:00
#12 1.014   Preparing metadata (setup.py): started
#12 1.128   Preparing metadata (setup.py): finished with status 'done'
#12 1.150 Collecting gym>=0.17.0
#12 1.186   Downloading gym-0.21.0.tar.gz (1.5 MB)
#12 1.221      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.5/1.5 MB 46.0 MB/s eta 0:00:00
#12 1.278   Preparing metadata (setup.py): started
#12 1.387   Preparing metadata (setup.py): finished with status 'done'
#12 1.532 Collecting numpy>=1.13.3
#12 1.565   Downloading numpy-1.22.2-cp38-cp38-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (13.4 MB)
#12 1.874      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 13.4/13.4 MB 45.6 MB/s eta 0:00:00
#12 1.903 Collecting cloudpickle>=1.3
#12 1.935   Downloading cloudpickle-2.0.0-py3-none-any.whl (25 kB)
#12 1.948 Collecting gin-config>=0.4.0
#12 1.981   Downloading gin_config-0.5.0-py3-none-any.whl (61 kB)
#12 1.982      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 61.3/61.3 KB 333.0 MB/s eta 0:00:00
#12 2.097 Collecting pillow
#12 2.137   Downloading Pillow-9.0.1-cp38-cp38-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (4.2 MB)
#12 2.220      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 4.2/4.2 MB 51.4 MB/s eta 0:00:00
#12 2.225 Requirement already satisfied: six>=1.10.0 in /usr/lib/python3/dist-packages (from tf-agents[reverb]) (1.14.0)
#12 2.241 Collecting absl-py>=0.6.1
#12 2.272   Downloading absl_py-1.0.0-py3-none-any.whl (126 kB)
#12 2.274      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 126.7/126.7 KB 369.2 MB/s eta 0:00:00
#12 2.383 Collecting protobuf>=3.11.3
#12 2.516   Downloading protobuf-3.19.4-cp38-cp38-manylinux2014_aarch64.whl (913 kB)
#12 2.540      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 913.6/913.6 KB 42.1 MB/s eta 0:00:00
#12 2.561 Collecting tensorflow-probability>=0.14.1
#12 2.768   Downloading tensorflow_probability-0.15.0-py2.py3-none-any.whl (5.7 MB)
#12 2.879      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 5.7/5.7 MB 52.7 MB/s eta 0:00:00
#12 2.902 Collecting typing-extensions>=3.7.4.3
#12 2.935   Downloading typing_extensions-4.0.1-py3-none-any.whl (22 kB)
#12 2.997 Collecting tf-agents[reverb]
#12 3.118   Downloading tf_agents-0.10.0-py3-none-any.whl (1.3 MB)
#12 3.159      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.3/1.3 MB 33.2 MB/s eta 0:00:00
#12 3.589   Downloading tf_agents-0.9.0-py3-none-any.whl (1.3 MB)
#12 3.634      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.3/1.3 MB 38.4 MB/s eta 0:00:00
#12 3.766   Downloading tf_agents-0.8.0-py3-none-any.whl (1.2 MB)
#12 3.796      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.2/1.2 MB 44.1 MB/s eta 0:00:00
#12 3.812 Collecting tensorflow-probability==0.12.2
#12 4.065   Downloading tensorflow_probability-0.12.2-py2.py3-none-any.whl (4.8 MB)
#12 4.461      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 4.8/4.8 MB 12.3 MB/s eta 0:00:00
#12 4.475 Collecting tf-agents[reverb]
#12 4.596   Downloading tf_agents-0.7.1-py3-none-any.whl (1.2 MB)
#12 4.617      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.2/1.2 MB 55.1 MB/s eta 0:00:00
#12 4.757   Downloading tf_agents-0.6.0-py3-none-any.whl (1.1 MB)
#12 4.781      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.1/1.1 MB 49.7 MB/s eta 0:00:00
#12 4.808 Collecting cloudpickle==1.3
#12 4.913   Downloading cloudpickle-1.3.0-py2.py3-none-any.whl (26 kB)
#12 4.920 Collecting tf-agents[reverb]
#12 5.046   Downloading tf_agents-0.5.0-py3-none-any.whl (933 kB)
#12 5.070      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 933.3/933.3 KB 51.4 MB/s eta 0:00:00
#12 5.080 WARNING: tf-agents 0.5.0 does not provide the extra 'reverb'
#12 5.084 Collecting gin-config==0.1.3
#12 5.188   Downloading gin_config-0.1.3-py3-none-any.whl (43 kB)
#12 5.191      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 43.3/43.3 KB 187.0 MB/s eta 0:00:00
#12 5.225 Collecting decorator
#12 5.264   Downloading decorator-5.1.1-py3-none-any.whl (9.1 kB)
#12 5.281 Collecting gast>=0.3.2
#12 5.313   Downloading gast-0.5.3-py3-none-any.whl (19 kB)
#12 5.332 Collecting dm-tree
#12 5.437   Downloading dm-tree-0.1.6.tar.gz (33 kB)
#12 5.447   Preparing metadata (setup.py): started
#12 5.556   Preparing metadata (setup.py): finished with status 'done'
#12 5.563 Building wheels for collected packages: dm-tree
#12 5.564   Building wheel for dm-tree (setup.py): started
#12 5.671   Building wheel for dm-tree (setup.py): finished with status 'error'
#12 5.674   error: subprocess-exited-with-error
#12 5.674   
#12 5.674   × python setup.py bdist_wheel did not run successfully.
#12 5.674   │ exit code: 1
#12 5.674   ╰─> [5 lines of output]
#12 5.674       running bdist_wheel
#12 5.674       running build
#12 5.674       running build_py
#12 5.674       running build_ext
#12 5.674       error: command 'bazel' failed: No such file or directory
#12 5.674       [end of output]
#12 5.674   
#12 5.674   note: This error originates from a subprocess, and is likely not a problem with pip.
#12 5.675   ERROR: Failed building wheel for dm-tree
#12 5.675   Running setup.py clean for dm-tree
#12 5.770 Failed to build dm-tree
#12 5.806 Installing collected packages: protobuf, numpy, gin-config, gast, dm-tree, decorator, cloudpickle, absl-py, tensorflow-probability, tf-agents
#12 6.866   Running setup.py install for dm-tree: started
#12 6.965   Running setup.py install for dm-tree: finished with status 'error'
#12 6.967   error: subprocess-exited-with-error
#12 6.967   
#12 6.967   × Running setup.py install for dm-tree did not run successfully.
#12 6.967   │ exit code: 1
#12 6.967   ╰─> [7 lines of output]
#12 6.967       running install
#12 6.967       /usr/local/lib/python3.8/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
#12 6.967         warnings.warn(
#12 6.967       running build
#12 6.967       running build_py
#12 6.967       running build_ext
#12 6.967       error: command 'bazel' failed: No such file or directory
#12 6.967       [end of output]
#12 6.967   
#12 6.967   note: This error originates from a subprocess, and is likely not a problem with pip.
#12 6.968 error: legacy-install-failure
#12 6.968 
#12 6.968 × Encountered error while trying to install package.
#12 6.968 ╰─> dm-tree
#12 6.968 
#12 6.968 note: This is an issue with the package mentioned above, not pip.
#12 6.968 hint: See above for output from the failure.
------
executor failed running [/bin/sh -c $python_version -mpip --no-cache-dir install $tf_agents_version]: exit code: 1

```

Running Environment:

OS: Mac OS
Python Version: Python3 
Docker Version 20.10.12

pariskang, 2022-07-07:
how to resolve this error, I really need u help.I met it again.

shrutimary15, 2022-12-22:
I have also got same error

tfboyd, 2023-04-28:
I know this is a late response; but for anyone else coming. I just redid the Ariane Risc-V examples and also added a e2e smoke test that also lets you run GPU on a single machine just to verify the setup and have a baseline to work from. https://github.com/google-research/circuit_training/tree/main/tools#end-to-end-smoke-test

closed 2023-04-28 by tfboyd