google-research/circuit_training #11: Could you share the plc_wrapper_main

Yangxiaojun1230, 2022-02-18:
Thanks a lot for sharing such grate project!
I am digging into the sub-modules, could you share some informatiom about the plc? It seems like an EDA tool, could read def/lef, place cells ,handle blockages and so on. Is there any docs for it's api?

Vetelus, 2022-03-06:
I also found that "plc_wrapper_main" is missing in the project, so it seems that it is impossible to run the program normally. Have you found a solution?

esonghori, 2022-03-06:
Hi, 

The Placement Cost object is an approximation of what EDA tools are doing: placing std cells and QoR measurement (WL, Congestion, etc), but it is a lot quicker than EDA tools. You can find the API in [this test](https://github.com/google-research/circuit_training/blob/main/circuit_training/environment/plc_client_test.py).

The binary is stored at storage.googleapis.com, please follow [README](https://github.com/google-research/circuit_training/blob/main/README.md) on how to set up the project.
```
sudo curl https://storage.googleapis.com/rl-infra-public/circuit-training/placement_cost/plc_wrapper_main \
     -o  /usr/local/bin/plc_wrapper_main
```

Yangxiaojun1230, 2022-03-07:
> Hi,
>
> The Placement Cost object is an approximation of what EDA tools are doing: placing std cells and QoR measurement (WL, Congestion, etc), but it is a lot quicker than EDA tools. You can find the [...]

Thanks for your reply , esonghori
I could set up the project and I want to improve the WL and Congestion model or add some additional cost. So if you could share the source code, it will be easy to go on. 
By the way, could you also share your automatic scripts for transforming common verilog and lef files into initial.plc and netlist.pb.txt? I would like to test some other cases.

esonghori, 2022-03-20:
Unfortunately, the source code for the plc_wrapper_main binary includes lots of internal Google dependencies which make extremely hard to clean for open-sourcing.

For the lef/def to netlist.pb.txt, please see #3 and #10.

closed 2022-03-20 by esonghori