google-research/circuit_training #17: netlist.pb and plc files' format

wwrrzzttcc, 2022-04-02:
We are trying to write a tcl to do the transform from LEF/DEF to pb.netlist, so there are a few questions want to confirm first.
1. pb.netlist and its generated plc files are all the input needed?
2. node  rst_ni seems defined as a port and connected to several clusters (Grp_1204/Grp_1203...), but why clk_i not? example as:

  name: "clk_i"
  attr {
    key: "side"
    ...
node {
  name: "rst_ni"
  input: "Grp_1204/Pinput"
  input: "Grp_1203/Pinput"
  input: "Grp_791/Pinput"

Mr-Fang-VLSI, 2022-04-07:
Hi,

I think in one net, only one input drives multiple outputs. So I consider the pin with input as the driver. 

Donghao

wwrrzzttcc, 2022-04-07:
> Hi,
> 
> I think in one net, only one input drives multiple outputs. So I consider the pin with input as the driver.
> 
> Donghao

Hi  Fang,  I  believe we are facing some similar problems... so maybe lets have a chat more efficiently like by email..
<Ray.Wu@yhgch.com>

closed 2022-06-21 by sguada