google-research/circuit_training #20: How to generate .plc file

Mr-Fang-VLSI, 2022-04-07:
Hi, 

I have implemented the LEF/DEF converter, but I have some question on how to generate the .plc file.

1. how do we determine the density cost? 
2. how do we set the wirelength and congestion cost?
3. how do we set up the parameters about routers?
4. what is the meaning of smooth factor?

Thank you in advance!

wwrrzzttcc, 2022-04-07:
HI Fang
Great job！
I believe the wirelength cost is calculated by the location between two pins (or port) in canvas.  as the "HPWL" in the paper said.
BTW may u please share the converter (Ray.Wu@yhgch.com) because we are working on the same issue...
thanks so much..

Yangxiaojun1230, 2022-04-17:
> Hi,
>
> I have implemented the LEF/DEF converter, but I have some question on how to generate the .plc file.
>
> 1. how do we determine the density cost?
> 2. how do we set the wirelength and congestion [...]

Hi Fang，
I just had a look on your converter module. There are some questions to discuss:
1. Extract Macros from LEF file not all instance , since all instance called "macro" but the real macros should have the "CLASS = BLOCK" 
2. Dealing with standard cells, should use the tool hmetis to cluster into groups. And the information in graphdef should be the cluster instead of cells. ( When I deal with it , I don't know how to define the location of the cluster)

Btw, most functions are realized in python but have no dataset to verify it. If you have  a way, we can verify it. Thanks