google-research/circuit_training #27: plc_client.get_canvas_width_height unexplained behavior

Maria-UET, 2022-07-10:
Hi! 
Great work on circuit training. 

I came across this issue while parsing the [Ariane RISC netlist.pb.txt file ](https://github.com/google-research/circuit_training/blob/c25b27b240468f307c3cacef4efd9cf1db9248f2/circuit_training/environment/test_data/ariane/netlist.pb.txt)  with plc_client.  `plc_client.get_canvas_width_height()` gives the canvas size: (408.06, 408.06).  But I cannot find the source/logic behind this canvas size.
Note that:

- The netlist.pb.txt file has no explicit mention of canvas width and height. 

-  The [initial.plc](https://github.com/google-research/circuit_training/blob/c25b27b240468f307c3cacef4efd9cf1db9248f2/circuit_training/environment/test_data/ariane/initial.plc) file has a different size:  `# Width : 356.592  Height : 356.640` Also, I have tried changing the canvas size in initial.plc to random numbers but this does not affect the outcome of `plc_client.get_canvas_width_height()` proving that canvas size in initial.plc is not linked to canvas size from  `plc_client.get_canvas_width_height()` 

So, where does (408.06, 408.06) come from?

sakundu, 2022-09-02:
By default plc_client computes the total area of macros and soft macros and then create a square canvas with 60% utilization.

If you want to use different canvas size you have to add comment in your protobuf netlist or in the input plc file. You can see [placement_util.py](https://github.com/google-research/circuit_training/blob/d5e454e5bcd153a95d320f664af0d1b378aace7b/circuit_training/environment/placement_util.py#L132) can extract canvas size and other details.

-------------------------------------------------------------------------------------------------------------------------------
You may check out our [MacroPlacement](https://github.com/TILOS-AI-Institute/MacroPlacement) repo. 

We provide a [script](https://github.com/TILOS-AI-Institute/MacroPlacement/blob/main/Flows/util/gen_pb.tcl) to write out the protobuf netlist from the Innovus tool which can be used as input to the [CircuitTraining grouping](https://github.com/google-research/circuit_training/tree/main/circuit_training/grouping) code to generate the clustered netlist. 

Also, you can train circuit training using the testcases available in [MacroPlacement/Flows](https://github.com/TILOS-AI-Institute/MacroPlacement/tree/main/Flows)/\<Enablement\>/\<Design\>/netlist/output_CT_Grouping/.

If you are interested to learn more about [proxy cost](https://github.com/google-research/circuit_training/blob/d5e454e5bcd153a95d320f664af0d1b378aace7b/circuit_training/environment/environment.py#L60-L115) computation, you can visit this [link](https://docs.google.com/document/d/1hM7UbmANkhoGB3-UfFBp8TRDvvVjpmio7cyyjK4a5bI/edit).

You can find [here](https://docs.google.com/document/d/1HHZNcid5CZvvRqj_njzF7hBhtNSpmRn3fCYniWNYBiY/edit) the progress we have made till August 26.

jagdisht12817, 2023-06-30:
Hi
I have a doubt on canvas shape, will it always be a rectangular block? If not then how can we set the different shape