google-research/circuit_training #31: AttributeError: 'PPOLossInfo' object has no attribute 'clip_fraction'

ZFTurbo, 2022-07-29:
I have an error after building a docker. Tests were ok. Next I run example script:

```
... 

python3 -m circuit_training.learning.train_ppo \
  --root_dir=${ROOT_DIR} \
  --replay_buffer_server_address=${REVERB_SERVER} \
  --variable_container_server_address=${REVERB_SERVER} \
  --num_episodes_per_iteration=16 \
  --global_batch_size=64 \
  --netlist_file=${NETLIST_FILE} \
  --init_placement=${INIT_PLACEMENT}
  
  ...
```

Everything is fail at:

```
File "/usr/local/lib/python3.8/dist-packages/tensorflow/python/distribute/distribute_lib.py", line 1312, in run
    return self._extended.call_for_each_replica(fn, args=args, kwargs=kwargs)
  File "/usr/local/lib/python3.8/dist-packages/tensorflow/python/distribute/distribute_lib.py", line 2888, in call_for_each_replica
    return self._call_for_each_replica(fn, args, kwargs)
  File "/usr/local/lib/python3.8/dist-packages/tensorflow/python/distribute/distribute_lib.py", line 3689, in _call_for_each_replica
    return fn(*args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/tensorflow/python/autograph/impl/api.py", line 642, in wrapper
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/tf_agents/agents/tf_agent.py", line 336, in train
    loss_info = self._train_fn(
  File "/usr/local/lib/python3.8/dist-packages/tf_agents/utils/common.py", line 188, in with_check_resource_vars
    return fn(*fn_args, **fn_kwargs)
  File "/workspace/circuit_training/learning/agent.py", line 415, in _train
    data=loss_info.extra.clip_fraction,
AttributeError: 'PPOLossInfo' object has no attribute 'clip_fraction'
  In call to configurable 'Learner' (<class 'tf_agents.train.learner.Learner'>)
```

Probably we need to specify exact tf-agent version?

TF version in docker:

```
tensorflow==2.9.1
tensorflow-estimator==2.9.0
tensorflow-io-gcs-filesystem==0.26.0
tensorflow-probability==0.17.0
tf-agents==0.13.0
```

ZFTurbo, 2022-07-29:
I fixed this error by commenting this lines in `learning/agent.py` :
```
 	tf.compat.v2.summary.scalar(
	  name='clip_fraction',
	  data=loss_info.extra.clip_fraction,
	  step=self.train_step_counter)
	tf.compat.v2.summary.scalar(
	  name='grad_norm',
	  data=self._grad_norm,
	  step=self.train_step_counter)
```

Training is working now, but I'm not sure if correct.

roopaliv, 2022-07-29:
Hi Roman, 

Thanks for reaching out!

Based on the information provided in the issue, it looks like you recently pulled the latest version of circuit_training/ repository which updated the file circuit_training/learning/agent.py to the latest changes. However, these changes depend on latest changes in tf_agents repository. 

Therefore, you would need to update the  tf_agents repository as well so that you will get the latest changes in tf_agents/agents/ppo/ppo_agent.py. Once you update tf_agents/ppo/ppo_agent.py file, you will not need to comment the latest changes in learning/agent.py.

tfboyd, 2022-07-29:
@ZFTurbo   I need to run the test myself (I will try in the next 24 hours) but if you rebuild the docker with the command below it will use tf-agents-nightly and tensorflow-nightly which will have all the latest code from head. We run the same tests on it so it should be fine.  @roopaliv confirmed everything is checked in so it should all work.  

This is not tested but if you look at the [build](https://github.com/google-research/circuit_training/blob/main/tools/docker/ubuntu_circuit_training) file it looks correct. 

```shell
$ docker build --build-arg tf_agents_version=tf-agents-nightly[reverb] \
       --tag circuit_training:core -f tools/docker/ubuntu_circuit_training tools/docker/
```

Once I verify this I will add this to the README or atleast to the docker build file.

ZFTurbo, 2022-07-30:
There is also error:

```
File "/workspace/circuit_training/environment/environment.py", line 363, in CircuitEnv
 def call_analytical_placer_and_get_cost(self) -> tuple[float, InfoType]:
TypeError: 'type' object is not subscriptable
```

It was fixed by 

```
 def call_analytical_placer_and_get_cost(self) -> Tuple[float, InfoType]:
```

ZFTurbo, 2022-07-30:
May be it's good idea to update this line:
https://github.com/google-research/circuit_training/blob/main/tools/docker/ubuntu_circuit_training#L28

because tmux is not installed by default:

```
# Installs basics including add-apt.
RUN ${APT_COMMAND} update && ${APT_COMMAND} install -y --no-install-recommends \
        software-properties-common \
        curl \
	tmux \
        less
```

ZFTurbo, 2022-07-30:
> 

This worked for me, thanks. Training started after fixing 2 problems I wrote before.

tfboyd, 2022-08-01:
Awesome.  I will look at the tmux issue, document using tf-agents-nightly and take a look at your code tweak to see if we need to make a change.  Seems likely but that is not my expertise.  Thank you again.  I am running your command now so I can experience it while updating the docs.  I will link the PRs as I go and then close the issue.

tfboyd, 2022-08-02:
Alright I believe I have addressed all the issues:

- [Typing fixed](https://github.com/google-research/circuit_training/commit/867f60d33e68d35b4d5e780d649236cfcc08bc0b) We are also fixing this in other modules so the coded words in python less than 3.9.
- [Docker mentions](https://github.com/google-research/circuit_training/commit/2a193e3d6841b07fd9b8ae38bfc9cebad555a99f) how to use tf-agents-nightly and added tmux (only 1.2MB). I use tmux outside the docker but it could be useful in the docker so I added it for you. :-)

closed 2022-08-02 by tfboyd