google-research/circuit_training #32: Failed training for test_data: macro_tiles_10x10 and sample_clustered

samli50801, 2022-08-01:
We have trouble running these two testing data:  macro_tiles_10x10 and sample_clustered, although we succeeded in executing the training job for Ariane. The problem is that the tmux session for collect_job is stuck at iteration 0 and does not progress, which is show in the following figure.

[image]

We only changed the name of the directory in commands as follows:

$  export NETLIST_FILE=./circuit_training/environment/test_data/sample_clustered/netlist.pb.txt
$  export INIT_PLACEMENT=./circuit_training/environment/test_data/sample_clustered/initial.plc

1. Are we required to change other commands?

Maria-UET, 2022-08-12:
I also got stuck with model_id 0/step 0 when i trained for Ariane example AFTER changing the sizes of the macros randomly to bigger numbers. Can you compare the dimensions of macros in your netlist with those in Ariane?
May be it is the limitation on macro size in comparison to grid cell size.

samli50801, 2022-08-15:
We haven't found the clue between macro size and grid cell size yet. However, we found that model_id/step will stuck at 0 after we only modify the grid size from 35x33 to 35x32 in Ariane.
We are still looking for the reasons.

ZFTurbo, 2022-08-24:
I also wasn't able to run these 2 tests. Training stuck at iteration 0. And nothing is happening.

sakundu, 2022-09-02:
macro_tiles_10x10 has 100 macros. So you have to update the "sequence_length" to 101 in [train_ppo.py](https://github.com/google-research/circuit_training/blob/d5e454e5bcd153a95d320f664af0d1b378aace7b/circuit_training/learning/train_ppo.py#L57) and "max_sequence_length" to 101 in [ppo_collect.py](https://github.com/google-research/circuit_training/blob/d5e454e5bcd153a95d320f664af0d1b378aace7b/circuit_training/learning/ppo_collect.py#L53) before starting the training. 

"sequence_length" and "max_sequence_length" should be set to (TOTAL_NUMBER_OF_MACRO + 1) for training. The default value is 134 because the ariane test case has 133 macros.

-------------------------------------------------------------------------------------------------------------------------------
You may check out our [MacroPlacement](https://github.com/TILOS-AI-Institute/MacroPlacement) repo. 

We provide a [script](https://github.com/TILOS-AI-Institute/MacroPlacement/blob/main/Flows/util/gen_pb.tcl) to write out the protobuf netlist from the Innovus tool which can be used as input to the [CircuitTraining grouping](https://github.com/google-research/circuit_training/tree/main/circuit_training/grouping) code to generate the clustered netlist. 

Also, you can train circuit training using the testcases available in [MacroPlacement/Flows](https://github.com/TILOS-AI-Institute/MacroPlacement/tree/main/Flows)/\<Enablement\>/\<Design\>/netlist/output_CT_Grouping/.

If you are interested to learn more about [proxy cost](https://github.com/google-research/circuit_training/blob/d5e454e5bcd153a95d320f664af0d1b378aace7b/circuit_training/environment/environment.py#L60-L115) computation, you can visit this [link](https://docs.google.com/document/d/1hM7UbmANkhoGB3-UfFBp8TRDvvVjpmio7cyyjK4a5bI/edit).

You can find [here](https://docs.google.com/document/d/1HHZNcid5CZvvRqj_njzF7hBhtNSpmRn3fCYniWNYBiY/edit) the progress we have made till August 26.