google-research/circuit_training #34: cannot get hmetis

WONJUNYOO, 2022-08-10:
web site not working, so cannot get hmetis!
could you give hmetis binary file?

sclDL, 2022-08-15:
Hello,
I'm also tracking this issue! I found hmetis-1.5-linux gitlab archive from: https://gitlab.cecs.anu.edu.au/u5568237/cf2cs/tree/0f758d7b06c597e30db0579695107aadff0db486/hmetis-1.5-linux
I notice that the last gitlab update was about 5 years ago. I was wondering if you are using this version.

sclDL, 2022-08-15:
I am also experiencing this issue as well. Would you share the binary file for us? -- thanks!

esonghori, 2022-08-17:
Please download it from here: https://drive.google.com/file/d/1KfsYiQGp3PRlU1htXZ_G_VRDBETpPkdh/view?usp=sharing

closed 2022-08-17 by esonghori

carlosal1015, 2024-09-05:
https://karypis.github.io/glaros/files/sw/hmetis/hmetis-1.5-linux.tar.gz