google-research/circuit_training #37: Potential bug in getting blockages

Dinple, 2022-08-31:
Hi,

I think there is some minor issue with [get_blockages_from_comments](https://github.com/google-research/circuit_training/blob/83fe7210ba9518c467a59fa0eb5cd3bd6cace551/circuit_training/environment/placement_util.py#L106). Function header allows the filename to be either a string or a list of string, but the actual function definition treats string as a list nonetheless.

If you run placement_util_test.py, you will fail to extract blockage information and get something like this:
[image]

This is potentially an issue for [create_placement_cost_using_common_arguments](https://github.com/google-research/circuit_training/blob/83fe7210ba9518c467a59fa0eb5cd3bd6cace551/circuit_training/environment/placement_util.py#L751) since it passes a string to retrieve blockages. This function is also used in grouper.py.

Could you please confirm this? Thanks in advance!

esonghori, 2022-08-31:
Thanks for pointing this bug out, I'll send a fix shortly.

closed 2022-08-31 by esonghori

esonghori, 2022-09-01:
https://github.com/google-research/circuit_training/commit/d5e454e5bcd153a95d320f664af0d1b378aace7b should fix the issue.