google-research/circuit_training #39: How to load and use a pre-trained model (weights)?

RustamC, 2022-09-19:
Hi!

I've been playing around with circuit_training and now there's one question that's bothering me a lot.

Let's say, I trained model from scratch for Ariane. And now I want to use the obtained pre-trained weights for macro_tiles_10x10. What are my next steps?

I've tried to find any options that allow loading a pre-trained model, but I failed.

mumu157, 2023-06-12:
"Hello, have you solved it? Maybe you can get ideas from https://github.com/TILOS-AI-Institute/MacroPlacement. I would appreciate it if you could share them with me."

esonghori, 2023-06-12:
Please check this recent commit for how to load a trained checkpoint: https://github.com/google-research/circuit_training/commit/5cd4c5973f82480624eeec776a3c8a4b17b82e1b

closed 2023-06-12 by esonghori