google-research/circuit_training #41: Grouping no longer supports large protobuf netlist

sakundu, 2022-10-07:
Hi,
After this [commit](https://github.com/google-research/circuit_training/commit/d5e454e5bcd153a95d320f664af0d1b378aace7b), I see the grouping code no longer supports two smaller split protobuf netlists as input as mentioned [here](https://github.com/google-research/circuit_training/tree/main/circuit_training/grouping#note).  So I can not run the latest grouping code for the large protobuf netlist after splitting it into smaller netlist using [split_proto_netlist_main.py](https://github.com/google-research/circuit_training/blob/main/circuit_training/grouping/split_proto_netlist_main.py).

isu51004, 2025-01-13:
Hello. @sakundu 

Have you solved this problem? 

Thank you.

sakundu, 2025-01-13:
@isu51004 

I would suggest using that old commit (the one pointed above) to run grouping.

Thanks,
Sayak