google-research/circuit_training #46: Missing get_dreamplace_params function

sakundu, 2023-02-05:
Hi,
I have tried to run DREAMPlace to place the soft macros for the Ariane testcase available in the [test_data](https://github.com/google-research/circuit_training/tree/main/circuit_training/environment/test_data/ariane) directory using the following command:
```
python3 -m circuit_training.dreamplace.dreamplace_main --netlist_file=circuit_training/environment/test_data/ariane/netlist.pb.txt --init_placement=circuit_training/environment/test_data/ariane/initial.plc --output_dir=./output
```
The following screenshot shows the error message. Error message is **AttributeError: module 'dreamplace.Params' has no attribute 'get_dreamplace_params'**
[image]
  
I have used the [ubuntu_circuit_training](https://github.com/google-research/circuit_training/blob/main/tools/docker/ubuntu_circuit_training) docker build.

**Also, I have looked into the DREAMPlace [Params.py](https://github.com/limbo018/DREAMPlace/blob/master/dreamplace/Params.py) and I do not see any *get_dreamplace_params* function. So could you please also provide the changes required in DREAMPlace to make it work with CT?**

Thanks,
Sayak

esonghori, 2023-02-06:
Hi,

You should clone and install this branch
<https://github.com/limbo018/DREAMPlace/blob/circuit_training/dreamplace/Params.py>
of
the DREAMPlace. I'll change that in our code.

Thanks,
Ebrahim

sakundu, 2023-02-06:
Hi @esonghori 

Thank you for sharing the details. I have cloned the circuit_training branch of DREAMPlace and tried to run the following command. I have not received the previous error. But I am getting some new error. 
```
python3 -m circuit_training.dreamplace.dreamplace_main --netlist_file=./circuit_training/environment/test_data/ariane/netlist.pb.txt --init_placement=./circuit_training/environment/test_data/ariane/initial.plc --output_dir=./output
```

Here is the screenshot of the error message:
[image]

Here is the [log file](https://github.com/google-research/circuit_training/files/10668868/dreamplace_main.log)
) of the above run. Please let me know if I am doing something wrong and how to fix this problem.

Thanks again,
Sayak

esonghori, 2023-02-06:
The problem is the numpy float type.  Can you change this line to float64 and retry: https://github.com/google-research/circuit_training/blob/334646f4bc70b757243b4184f713f2532ac77f97/circuit_training/dreamplace/plc_converter.py#L467

sakundu, 2023-02-07:
Hi @esonghori 
I looked into the code and based on my understanding CT is not updating the DREAMPlace C++ database. So, I commented [this line](https://github.com/limbo018/DREAMPlace/blob/1e4503cd8d05969a162f9c293c2549ac7814b5df/dreamplace/PlaceDB.py#L1023) and now I can run the following command without any error. 

```
python3 -m circuit_training.dreamplace.dreamplace_main --netlist_file=./circuit_training/environment/test_data/ariane/netlist.pb.txt --init_placement=./circuit_training/environment/test_data/ariane/initial.plc --output_dir=./output
```

Thanks,
Sayak

closed 2023-02-07 by sakundu