google-research/circuit_training #47: Configurations for running on GPU?

Mohan2351999, 2023-02-06:
I'm actually trying to run the process on single GPU with 32GB gives me out of memory issue, 

If I'm using multiple GPUs, how do I execute the training process? All the jobs in training are inter-related right? Can I assign a different GPU for a different process using CUDA_VISIBLE_DEVICES?

Can I directly use the code for CPU version, or do I have to make changes to execute this on the GPU environment.

Any help is highly appreciated,

Thanks,
Mohan.

tfboyd, 2023-04-28:
This is a very late response. I have updated and created and e2e example that runs training on GPU and everything else on CPU.  https://github.com/google-research/circuit_training/tree/main/tools#end-to-end-smoke-test

closed 2023-04-28 by tfboyd

RaidToRadar, 2023-05-30:
> This is a very late response. I have updated and created and e2e example that runs training on GPU and everything else on CPU. https://github.com/google-research/circuit_training/tree/main/tools#end-to-end-smoke-test

The e2e example for GPU does not work due to the branch version. The ubuuntu_circuit_training script is not updated in r.0.0.3 making not even be ran. From my understanding GPU is only supported on the devel/HEAD branch but then the code doesn't really work for that

I am just trying to get the e2e_smoke test to run using the GPU, perhaps the code requires some tweaks or clarification in that regard.