google-research/circuit_training #48: ppo_train is erroring out when dreamplace_core is imported

sakundu, 2023-02-11:
Hi,

I am trying run CT with dreamplace and I am getting the following error from the [training log](https://github.com/google-research/circuit_training/files/10713859/ct_train.log).
```
W tensorflow/core/framework/op_kernel.cc:1830] OP_REQUIRES failed at matmul_op_impl.h:622 : INTERNAL: cublas error
```
Here is the screenshot of the error:
[image]

So I tried running CT using FD and I was still getting the same error.  Then I observed if I do not import dreamplace_core I do not have any such problem.

Can you please help me to figure out why I am getting this error? Also, how to fix it?

Thanks,
Sayak

esonghori, 2023-02-12:
I am guessing the problem is due to sharing GPU between DREAMPlace and the PPO training. Can you make sure that DREAMPlace runs on CPU and not GPU?

sakundu, 2023-02-13:
Hi @esonghori 

I compiled the DREAMPlace without enabling the GPU and I was still getting the error. I was running it on my local environment. I have started using the docker environment and I am not getting this error any more. So, I am assuming there is some  problem in my local environment. 

Thanks,
Sayak