google-research/circuit_training #49: Type annotation, variable length mismatch error while running the latest CircuitTraining

sakundu, 2023-02-13:
Hi
I was running a old release of Circuit Training (CT) (commit tag: 91e14fd1caa5b15d9bb1b58b6d5e47042ab244f3). I have integrated the DREAMPlace in it and observed my runs are running out of memory after ~50 iteration of training.

I am trying to run the latest release of CT. Here are the details of my environment:
- I have created my own DREAMPlace build using [bootstrap_dreamplace_build.sh](https://github.com/google-research/circuit_training/blob/main/tools/bootstrap_dreamplace_build.sh) for python3.8.
- For DREAMPlace I am using the [circuit training branch](https://github.com/limbo018/DREAMPlace/tree/circuit_training).
- I have created the docker environment using [ubuntu_circuit_training](https://github.com/google-research/circuit_training/blob/main/tools/docker/ubuntu_circuit_training) and used **nvidia/cuda:11.4.2-cudnn8-devel-ubuntu20.04** as base image. Changed the default numpy version from 1.24.2 to 1.23.1.

I am getting the following errors:
- Lots of type annotation error. Link to one of [error](https://github.com/google-research/circuit_training/blob/4195b6b5c1e4b7eeb084976f2332bbfdda85d12b/circuit_training/dreamplace/dreamplace_util.py#L67). There are many more in [coordinate_descent_placer.py](https://github.com/google-research/circuit_training/blob/main/circuit_training/environment/coordinate_descent_placer.py) and [model_lib.py](https://github.com/google-research/circuit_training/blob/main/circuit_training/model/model_lib.py). **Are these errors real or  am I supposed to use python 3.11?**
- I see ppo_train does not suppose *--num_episodes_per_iteration=16* any more.  It is there in the [README](https://github.com/google-research/circuit_training/blob/main/README.md).
- Shape mismatch error in eval_job and train_job after removing the *num_episode_per_iteration*. I have attached the [eval.log](https://github.com/google-research/circuit_training/files/10727144/eval.log) and [train.log](https://github.com/google-research/circuit_training/files/10727139/train.log) files.

So could you please suggest me is there any way to ensure that  I am using the correct environment even after using docker?

I tried to run ```python3 -m circuit_training.environment.environment_test``` and I found this error (after fixing the type annotation errors): **ERROR: test_save_file_train_step (__main__.EnvironmentTest)**

So could you please suggest if it is an environment problem at my end or it is some bug in the latest release of CT?

Thanks for your help,
Sayak

joewjiang, 2023-03-07:
Regarding the `--num_episodes_per_iteration` flag, it was removed and replaced by a train() function argument. You can pass it to the binary as a gin binding, such as `--gin_bindings='train.num_episodes_per_iteration=16'`

sakundu, 2023-03-07:
Hi @joewjiang 

Thank you for the examples.

I tried to run the e2e_smoke_test following the steps given in the [readme](https://github.com/google-research/circuit_training/tree/main/tools#tools-for-circuit-training). I found that after the first training iteration collector jobs are errored out due to "Infeasible action".

I have also attached the log file : [collect_1.log](https://github.com/google-research/circuit_training/files/10906820/collect_1.log).

Could you kindly help me to fix this?

Thanks again for your help,
Sayak

sakundu, 2023-05-01:
I no longer encounter these errors when using Python 3.9.

closed 2023-05-01 by sakundu