google-research/circuit_training #50: Request to provide a GPU-Accelerated Base Docker Image for CT

sakundu, 2023-04-07:
Hi,

When I use the ```base_image=nvidia/cuda:11.4.2-cudnn8-devel-ubuntu20.04``` to build the docker environment following the steps given [here](https://github.com/google-research/circuit_training/blob/691105687d0cef453467e073bfd0843725302cb8/tools/docker/ubuntu_circuit_training#L16-L17), I encounter the subsequent error:
```
2023-04-03 17:42:04.452598: E tensorflow/compiler/xla/stream_executor/cuda/cuda_dnn.cc:417] Loaded runtime CuDNN library: 8.5.0 but source was compiled with: 8.6.0.  CuDNN library needs to have matching major version and equal or higher minor version. If using a binary install, upgrade your CuDNN library.  If building from sources, make sure the library loaded at runtime is compatible with the version specified during compile configuration.
2023-04-03 17:42:04.455368: W tensorflow/core/framework/op_kernel.cc:1830] OP_REQUIRES failed at conv_grad_input_ops.cc:385 : UNIMPLEMENTED: DNN library is not found.
```

 **Could you kindly recommend a suitable base_image that is compatible with tf-agent[reverb] v0.16?**

Thanks,
Sayak

tfboyd, 2023-04-15:
Sorry I have not run with GPU in a while.  I might soon.  Based on the error the cuDNN in the docker is too older 8.5 and you need 8.6.  My guess is you should be able to fix that by changing the base image to something newer that has cuDNN 8.6.

tfboyd, 2023-04-15:
I did not test it; but this would seem to be a decent guess.
[11.8.0-cudnn8-devel-ubuntu22.04](https://hub.docker.com/layers/nvidia/cuda/11.8.0-cudnn8-devel-ubuntu22.04/images/sha256-5079d0d0f36cc63050a0f5c010d769c68b8e959c2d2a45f8ec44dd7e5c1bf7f9?context=explore)

tfboyd, 2023-04-28:
I have fixed the smoke tests and gotten GPU acceleration working in docker. There were some issues with torch needed by dreamplace; but that is fixed.  https://github.com/google-research/circuit_training/tree/main/tools#end-to-end-smoke-test

closed 2023-04-28 by tfboyd