google-research/circuit_training #52: Docker smoke test test_save_file_train_step fails: 'PlaceDB' object has no attribute 'flatten_nested_map'

abettati, 2023-04-12:
Here is the log out of the smoke test:
```
root@882a26d2ae64:/workspace# python3.9 -m circuit_training.environment.environment_test
2023-04-12 02:40:09.994638: I tensorflow/core/util/port.cc:110] oneDNN custom operations are on. You may see slightly different numerical results due to floating-point round-off errors from different computation orders. To turn them off, set the environment variable `TF_ENABLE_ONEDNN_OPTS=0`.
2023-04-12 02:40:10.246438: I tensorflow/tsl/cuda/cudart_stub.cc:28] Could not find cuda drivers on your machine, GPU will not be used.
2023-04-12 02:40:11.468534: I tensorflow/tsl/cuda/cudart_stub.cc:28] Could not find cuda drivers on your machine, GPU will not be used.
2023-04-12 02:40:11.475281: I tensorflow/core/platform/cpu_feature_guard.cc:182] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 AVX512F AVX512_VNNI FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
2023-04-12 02:40:14.729153: W tensorflow/compiler/tf2tensorrt/utils/py_utils.cc:38] TF-TRT Warning: Could not find TensorRT
/usr/lib/python3/dist-packages/requests/__init__.py:89: RequestsDependencyWarning: urllib3 (1.25.8) or chardet (5.1.0) doesn't match a supported version!
  warnings.warn("urllib3 ({}) or chardet ({}) doesn't match a supported "
Running tests under Python 3.9.16: /usr/bin/python3.9
[ RUN      ] EnvironmentTest.test_action_space
I0412 02:40:28.219354 139657645123392 environment.py:216] ***Num node to place***:2
I0412 02:40:28.223868 139657645123392 placement_util.py:447] node_order: descending_size_macro_first
I0412 02:40:28.245094 139657645123392 observation_extractor.py:320] Pad a tensor with shape (5,) by 42000
I0412 02:40:28.255223 139657645123392 observation_extractor.py:334] Pad a tensor with shape (5,) by 3500
INFO:tensorflow:time(__main__.EnvironmentTest.test_action_space): 0.34s
I0412 02:40:28.288695 139657645123392 test_util.py:2467] time(__main__.EnvironmentTest.test_action_space): 0.34s
[       OK ] EnvironmentTest.test_action_space
[ RUN      ] EnvironmentTest.test_create_and_obs_space
I0412 02:40:28.368595 139657645123392 environment.py:216] ***Num node to place***:2
I0412 02:40:28.370536 139657645123392 placement_util.py:447] node_order: descending_size_macro_first
I0412 02:40:28.376635 139657645123392 observation_extractor.py:320] Pad a tensor with shape (5,) by 42000
I0412 02:40:28.377619 139657645123392 observation_extractor.py:334] Pad a tensor with shape (5,) by 3500
INFO:tensorflow:time(__main__.EnvironmentTest.test_create_and_obs_space): 0.11s
I0412 02:40:28.401192 139657645123392 test_util.py:2467] time(__main__.EnvironmentTest.test_create_and_obs_space): 0.11s
[       OK ] EnvironmentTest.test_create_and_obs_space
[ RUN      ] EnvironmentTest.test_infisible
I0412 02:40:28.482485 139657645123392 environment.py:216] ***Num node to place***:2
I0412 02:40:28.485958 139657645123392 placement_util.py:447] node_order: descending_size_macro_first
I0412 02:40:28.503776 139657645123392 observation_extractor.py:320] Pad a tensor with shape (5,) by 42000
I0412 02:40:28.505141 139657645123392 observation_extractor.py:334] Pad a tensor with shape (5,) by 3500
INFO:tensorflow:time(__main__.EnvironmentTest.test_infisible): 0.12s
I0412 02:40:28.524590 139657645123392 test_util.py:2467] time(__main__.EnvironmentTest.test_infisible): 0.12s
[       OK ] EnvironmentTest.test_infisible
[ RUN      ] EnvironmentTest.test_save_file_train_step
I0412 02:40:28.815208 139657645123392 environment.py:216] ***Num node to place***:2
I0412 02:40:28.817175 139657645123392 placement_util.py:447] node_order: descending_size_macro_first
I0412 02:40:28.824875 139657645123392 observation_extractor.py:320] Pad a tensor with shape (5,) by 42000
I0412 02:40:28.828542 139657645123392 observation_extractor.py:334] Pad a tensor with shape (5,) by 3500
I0412 02:40:28.889569 139657645123392 placement_util.py:447] node_order: random
I0412 02:40:28.895797 139657645123392 dreamplace_util.py:104] Update num_bins_x and num_bins_y: (128, 128)
I0412 02:40:28.902512 139657645123392 plc_converter.py:141] Node 8 is placed at (170.000000, 230.000000).
I0412 02:40:28.907457 139657645123392 plc_converter.py:141] Node 2 is placed at (375.000000, 375.000000).
I0412 02:40:28.909322 139657645123392 plc_converter.py:141] Node 3 is placed at (125.000000, 125.000000).
I0412 02:40:28.911084 139657645123392 plc_converter.py:141] Node 0 is placed at (0.000000, 100.000000).
I0412 02:40:28.912482 139657645123392 plc_converter.py:141] Node 1 is placed at (499.000000, 499.000000).
INFO:tensorflow:time(__main__.EnvironmentTest.test_save_file_train_step): 0.4s
I0412 02:40:28.929174 139657645123392 test_util.py:2467] time(__main__.EnvironmentTest.test_save_file_train_step): 0.4s
[  FAILED  ] EnvironmentTest.test_save_file_train_step
[ RUN      ] EnvironmentTest.test_session
[  SKIPPED ] EnvironmentTest.test_session
[ RUN      ] EnvironmentTest.test_validate_circuite_env
I0412 02:40:29.045726 139657645123392 environment.py:216] ***Num node to place***:2
I0412 02:40:29.048648 139657645123392 placement_util.py:447] node_order: descending_size_macro_first
I0412 02:40:29.060537 139657645123392 observation_extractor.py:320] Pad a tensor with shape (5,) by 42000
I0412 02:40:29.064798 139657645123392 observation_extractor.py:334] Pad a tensor with shape (5,) by 3500
INFO:tensorflow:time(__main__.EnvironmentTest.test_validate_circuite_env): 0.25s
I0412 02:40:29.188475 139657645123392 test_util.py:2467] time(__main__.EnvironmentTest.test_validate_circuite_env): 0.25s
[       OK ] EnvironmentTest.test_validate_circuite_env
[ RUN      ] EnvironmentTest.test_wrap_tfpy_environment
I0412 02:40:29.263710 139657645123392 environment.py:216] ***Num node to place***:2
I0412 02:40:29.265571 139657645123392 placement_util.py:447] node_order: descending_size_macro_first
I0412 02:40:29.270780 139657645123392 observation_extractor.py:320] Pad a tensor with shape (5,) by 42000
I0412 02:40:29.271743 139657645123392 observation_extractor.py:334] Pad a tensor with shape (5,) by 3500
/usr/lib/python3.9/multiprocessing/pool.py:265: ResourceWarning: unclosed running multiprocessing pool <multiprocessing.pool.ThreadPool state=RUN pool_size=1>
  _warn(f"unclosed running multiprocessing pool {self!r}",
ResourceWarning: Enable tracemalloc to get the object allocation traceback
INFO:tensorflow:time(__main__.EnvironmentTest.test_wrap_tfpy_environment): 0.13s
[... 19 lines cut ...]
  File "/usr/local/lib/python3.9/dist-packages/gin/utils.py", line 41, in augment_exception_message_and_reraise
    raise proxy.with_traceback(exception.__traceback__) from None
  File "/usr/local/lib/python3.9/dist-packages/gin/config.py", line 1582, in gin_wrapper
    return fn(*new_args, **new_kwargs)
  File "/workspace/circuit_training/environment/coordinate_descent_placer.py", line 149, in __init__
    self._dreamplace = dreamplace_core.SoftMacroPlacer(
  File "/usr/local/lib/python3.9/dist-packages/gin/config.py", line 1605, in gin_wrapper
    utils.augment_exception_message_and_reraise(e, err_str)
  File "/usr/local/lib/python3.9/dist-packages/gin/utils.py", line 41, in augment_exception_message_and_reraise
    raise proxy.with_traceback(exception.__traceback__) from None
  File "/usr/local/lib/python3.9/dist-packages/gin/config.py", line 1582, in gin_wrapper
    return fn(*new_args, **new_kwargs)
  File "/workspace/circuit_training/dreamplace/dreamplace_core.py", line 34, in __init__
    self.placedb_plc = placedb_plc.PlacedbPlc(plc, params, hard_macro_order)
  File "/workspace/circuit_training/dreamplace/placedb_plc.py", line 31, in __init__
    self.placedb = self.converter.convert(plc, hard_macro_order)
  File "/workspace/circuit_training/dreamplace/plc_converter.py", line 524, in convert
    convert_to_ndarray(db)
  File "/workspace/circuit_training/dreamplace/plc_converter.py", line 404, in convert_to_ndarray
    db.flat_node2pin_map, db.flat_node2pin_start_map = db.flatten_nested_map(
AttributeError: 'PlaceDB' object has no attribute 'flatten_nested_map'
  In call to configurable 'SoftMacroPlacer' (<class 'circuit_training.dreamplace.dreamplace_core.SoftMacroPlacer'>)
  In call to configurable 'CoordinateDescentPlacer' (<class 'circuit_training.environment.coordinate_descent_placer.CoordinateDescentPlacer'>)
```
Looks like it is calling a wrong version of DREAMPlace?

esonghori, 2023-04-12:
Please use the circuit_training branch of DREAMPlace instead of the master branch: https://github.com/limbo018/DREAMPlace/blob/1e4503cd8d05969a162f9c293c2549ac7814b5df/dreamplace/PlaceDB.py#L445

closed 2023-04-12 by esonghori

abettati, 2023-04-13:
Hi @esonghori thanks for the reply.
I would like to understand better what did go wrong in my docker setup so that I can fix.
I was installing CT by following the main readme and I got redirected to a docker file

"DREAMPlace is not provided as a PyPi package and requires compiled code. The best option is to use the [docker image](https://github.com/google-research/circuit_training/blob/main/tools/docker/ubuntu_circuit_training)." 

So from there I followed the commands in the docker file and got the above error. Can you help me fix the procedure? 

Thank you

esonghori, 2023-04-13:
We are working to update the dreamplace images so they use the right branch.

tfboyd, 2023-04-15:
I just pushed new DREAMPlace binaries that I tested with the end-to-end script AND I added forcing DreamPlace to be used in the end-to-end test. Thank you for finding this and causing us to revisit the integration and improve testing. https://github.com/google-research/circuit_training/tree/main/tools#end-to-end-smoke-test