google-research/circuit_training #55: How do you generate .plc file using LEF/DEF translator?

RaidToRadar, 2023-06-12:
I have been attempting to use the LEF/DEF translator to use in circuit training, but the script only generates the .pb.txt file upon completion with the example that this given. From my understanding, the circuit training requires both the .plc and the .pb.txt file to work unless otherwise stated. I would really appreciate some assistance/guidance for this.

esonghori, 2023-06-12:
`grouper_main` doesn't need a plc file as an input. It only needs the pb.txt file. It will generate the plc file for the clustered netlist which you can use as an input to the RL algorithm.

closed 2023-06-12 by esonghori

RaidToRadar, 2023-06-13:
I have issues running grouper_main, mostly when attempting to import any module that has circuit_training, I have tried to use sys.path.append() but that means every script that I need to import circuit training will also require this change. I will try to run using python instead of python3 for now and see if that fixes the issue