google-research/circuit_training #59: CT not working in ubuntu22.04 VM

RaidToRadar, 2023-07-25:
I have a VM setup for running circuit training and while I was able to get the grouping code to work correctly, when I went to start the e2e_smoke test, after a few minutes, I received an error message about SoftMacroPlacer() timing out.

```
0724 19:54:09.211993 140645097709568 NonLinearPlace.py:318] iteration  124, ( 124,  0,  0), Obj 2.997260E+03, DensityWeight 1.436298E-06, HPWL 1.045501E+05, Overflow 9.715901E-01, MaxDensity 2.036E+02, gamma 1.922252E+02, time 241.354ms
I0724 19:54:09.215112 140645097709568 NonLinearPlace.py:327] full step 2189.637 ms
Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/workspace/circuit_training/learning/train_ppo.py", line 375, in <module>
    multiprocessing.handle_main(functools.partial(app.run, main))
  File "/usr/local/lib/python3.9/dist-packages/tf_agents/system/default/multiprocessing_core.py", line 77, in handle_main
    return app.run(parent_main_fn, *args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/absl/app.py", line 308, in run
    _run_main(main, args)
  File "/usr/local/lib/python3.9/dist-packages/absl/app.py", line 254, in _run_main
    sys.exit(main(argv))
  File "/usr/local/lib/python3.9/dist-packages/absl/app.py", line 308, in run
    _run_main(main, args)
  File "/usr/local/lib/python3.9/dist-packages/absl/app.py", line 254, in _run_main
    sys.exit(main(argv))
  File "/workspace/circuit_training/learning/train_ppo.py", line 327, in main
    env = create_env_fn()
  File "/workspace/circuit_training/environment/environment.py", line 629, in create_circuit_environment
    env = CircuitEnv(*args, **kwarg)
  File "/usr/local/lib/python3.9/dist-packages/gin/config.py", line 1605, in gin_wrapper
    utils.augment_exception_message_and_reraise(e, err_str)
  File "/usr/local/lib/python3.9/dist-packages/gin/utils.py", line 41, in augment_exception_message_and_reraise
    raise proxy.with_traceback(exception.__traceback__) from None
  File "/usr/local/lib/python3.9/dist-packages/gin/config.py", line 1582, in gin_wrapper
    return fn(*new_args, **new_kwargs)
  File "/workspace/circuit_training/environment/environment.py", line 285, in __init__
    converged = self._dreamplace.place()
  File "/workspace/circuit_training/dreamplace/dreamplace_core.py", line 65, in place
    return decorated_place()
  File "/usr/local/lib/python3.9/dist-packages/timeout_decorator/timeout_decorator.py", line 82, in new_function
    return function(*args, **kwargs)
  File "/workspace/circuit_training/dreamplace/dreamplace_core.py", line 62, in decorated_place
    return self._place()
  File "/workspace/circuit_training/dreamplace/dreamplace_core.py", line 49, in _place
    metrics = nonlinear_place(self.params, self.placedb_plc.placedb)
  File "/dreamplace/dreamplace/NonLinearPlace.py", line 448, in __call__
    one_descent_step(
  File "/dreamplace/dreamplace/NonLinearPlace.py", line 309, in one_descent_step
    optimizer.step()
  File "/usr/local/lib/python3.9/dist-packages/torch/optim/optimizer.py", line 140, in wrapper
    out = func(*args, **kwargs)
  File "/dreamplace/dreamplace/NesterovAcceleratedGradientOptimizer.py", line 121, in step
    f_kp1, g_kp1 = obj_and_grad_fn(v_kp1)
  File "/dreamplace/dreamplace/PlaceObj.py", line 387, in obj_and_grad_fn
    obj = self.obj_fn(pos)
  File "/dreamplace/dreamplace/PlaceObj.py", line 287, in obj_fn
    self.density = self.op_collections.density_op(pos)
  File "/usr/local/lib/python3.9/dist-packages/torch/nn/modules/module.py", line 1194, in _call_impl
    return forward_call(*input, **kwargs)
  File "/dreamplace/dreamplace/ops/electric_potential/electric_potential.py", line 521, in forward
    return ElectricPotentialFunction.apply(
  File "/dreamplace/dreamplace/ops/electric_potential/electric_potential.py", line 96, in forward
    density_map = ElectricDensityMapFunction.forward(
  File "/dreamplace/dreamplace/ops/electric_potential/electric_overflow.py", line 98, in forward
    output = electric_potential_cpp.density_map(
  File "/usr/local/lib/python3.9/dist-packages/timeout_decorator/timeout_decorator.py", line 69, in handler
    _raise_exception(timeout_exception, exception_message)
  File "/usr/local/lib/python3.9/dist-packages/timeout_decorator/timeout_decorator.py", line 47, in _raise_exception
    raise exception(exception_message)
timeout_decorator.timeout_decorator.TimeoutError: 'SoftMacroPlacer place() timed out.'
  In call to configurable 'CircuitEnv' (<class 'circuit_training.environment.environment.CircuitEnv'>)
Collect job failed (SIGUSR1). Check /workspace/logs/test/collect_*.log.
Exiting with code 8.
```

esonghori, 2023-07-25:
This can be a transient issue and it is there to detect DREAMPlace occasional deadlocks. If it persists, you can disable the timeout by  setting `enable_timeout` to false [here](https://github.com/google-research/circuit_training/blob/5c3b9e166fd19c0b72879e589503d6a52d66e65f/circuit_training/dreamplace/dreamplace_core.py#L31).

RaidToRadar, 2023-07-27:
That seemed to fix the issue on my end, thank you very much.

closed 2023-07-27 by RaidToRadar

samli50801, 2024-10-30:
> That seemed to fix the issue on my end, thank you very much.
Hi, I set the ```enable_timeout``` to false and the timeout issue are solved. But the training job is then stuck at 
```I tensorflow/core/kernels/data/shuffle_dataset_op.cc:452] Shuffle buffer filled.```
Did you face this problem as well? Thank you